qiayuanl/legged_control #1: quad_gazebo build error

littleggghost, 2022-07-05:
I use `catkin build` to compile quad_gazebo, and meet some protobuf related error.
The errors are so long and can't find the beginning in terminal.

my env: ubuntu20.04+ros noetic+gazebo11.10

littleggghost, 2022-07-11:
Did your  environment also: ubuntu20.04+ros noetic+gazebo11.10 ?
I'm sure this is an environment releated issue.

qiayuanliao, 2022-07-11:
> Did your environment also: ubuntu20.04+ros noetic+gazebo11.10 ? I'm sure this is an environment releated issue.

Yes, I am in the same environment. Some people in the same environment run these packages successfully.

1874ruanshaodong, 2022-07-28:
i run this code in ubuntu18.04+ros melodic, and it makes error, can't compile successfully.first,it publish"CMake Error at legged_control/qpoases_catkin/CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.14 or higher is required.  You are running version 3.10.2
", secondly, i rewrite the required 3.10, but if publish another error, "CMake Error at legged_control/qpoases_catkin/CMakeLists.txt:6 (include):include could not find load file:FetchContent 
CMake Error at legged_control/qpoases_catkin/CMakeLists.txt:26 (FetchContent_Declare):
  Unknown CMake command "FetchContent_Declare".", 
in the end, i will try it in ubuntu20.04.

qiayuanliao, 2022-07-28:
> 

Ubuntu 18.04 is not supported by OCS2

closed 2022-07-28 by qiayuanliao