qiayuanl/legged_control #2: How do you apply ocs2 to the real AlienGo?

skywoodsz, 2022-07-31:
Hi, it's an amazing project. OCS2 is a useful mpc tool, but it requires ubuntu2004. I tried to run it on ubuntu1604 but failed. 
It looks like OCS2 only works on ubuntu2004. However, the mini-computer system on AlienGo is ubuntu1604. 
May I ask how you run your project on the real AlienGo? Can you give me some suggestions？
Thanks!

qiayuanliao, 2022-07-31:
Whether it is in A1 or AlienGo, you only need to connect the computer(NUC or your laptop) through a network cable and configure the computer IP to 192.168.123.1XX, you don't need to care about the native computer on the robot itself.

skywoodsz, 2022-08-02:
Thanks for your advice. I will give you feedback after trying it. Thanks!

closed 2022-08-02 by skywoodsz