qiayuanl/legged_control #5: A1 can't stand and something wrong with rviz

P1terQ, 2022-09-24:
I follow the order : 1.`roslaunch unitree_description empty_world.launch ` 2. `roslaunch legged_controllers load_controller.launch cheater:=true` 3.`rosrun rqt_controller_manager rqt_controller_manager` 4.start the legged_controller option in rqt_controller_manager 5.enter the gait . **The motors move in the gait i entered but the robot keep lie on the ground**   When i use `rviz`, there is no robot or  scroll bar, and it give an error in the Global Status: **unknown frame map**. How can I fix this problem?

qiayuanliao, 2022-09-24:
Set fixed frame of RViz to odom.

P1terQ, 2022-09-24:
the error in rviz disappeared, but still no robot and scroll bar. I notice there is a warn in the `roslaunch unitree_description empty_world.launch` window, 
`[ WARN] [1664017856.345256560, 10.290000000]: The root link base has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.`
Is it normal?

qiayuanliao, 2022-09-24:
> the error in rviz disappeared, but still no robot and scroll bar. I notice there is a warn in the `roslaunch unitree_description empty_world.launch` window,
> `[ WARN] [1664017856.345256560, [...]

Yes, it's normal.

qiayuanliao, 2022-09-24:
> the error in rviz disappeared, but still no robot and scroll bar. I notice there is a warn in the `roslaunch unitree_description empty_world.launch` window,
> `[ WARN] [1664017856.345256560, [...]

The robot and scroll bar need to be added by yourself, you should learn how to use Rviz and rqt. This is some basics of ROS, I also suggest you should learn the brief introduction tutorial of ROS before play with this project.

closed 2022-09-24 by P1terQ