qiayuanl/legged_control #8: The problem that the simulator cannot be loaded when loading the go1 model

suihu-mk, 2022-09-30:
I have a problem when trying to modify a1 model to go1 model，[ERROR] [1664549659.767555470, 0.166000000]: **Skipped loading plugin with error: XML Document '/legged_hw_sim_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..**
[FATAL] [1664549659.767876144, 0.166000000]: Failed to create robot simulation interface loader: **According to the loaded plugin descriptions the class legged_gazebo/QuadHWSim with base class type gazebo_ros_control::RobotHWSim does not exist.** Declared types are  gazebo_ros_control/DefaultRobotHWSim
****
I have modified the xacro file and urdf file, the model has been successfully displayed in gezebo, but the above error is prompted.I would like to inquire about how to configure the legged_hw_sim_plugins.xml. Can you give me some advice? Thanks！

qiayuanliao, 2022-09-30:
Can you run a1 without error?

If true, I am sure that you didn't need to modify legged_hw_sim_plugins.xml. I guess there is some problem in your go1 URDF.

suihu-mk, 2022-09-30:
There is no error in running a1. Here is the urdf file of go1. Can you help me check it?
[go1.txt](https://github.com/qiayuanliao/legged_control/files/9685919/go1.txt)

suihu-mk, 2022-10-01:
I add the go1 model file in the unitree_description/meshes/go1/ folder, and place the const.xacro file in the unitree_description/urdf/go1  folder. I do not modify the robot.xacro file, and then set the ROBOT_TYPE=go1 on the terminal.  
The simulation was run successfully.
When I modify the robot.xacro file and generate the urdf file, it cannot run successfully.
I don't know if there is another way.
And，if I want to deploy to the real go1, what should I adjust.

suihu-mk, 2022-10-01:
[image]
[image]

qiayuanliao, 2022-10-01:
> There is no error in running a1. Here is the urdf file of go1. Can you help me check it? [go1.txt](https://github.com/qiayuanliao/legged_control/files/9685919/go1.txt)

Don't send me the generated URDF file.

qiayuanliao, 2022-10-01:
> When I modify the robot.xacro file and generate the urdf file, it cannot run successfully.
What kind of modify?

qiayuanliao, 2022-10-01:
> [image] [image]

The simulation launch file requires a ".xarco" file, and the OCS2 requires a ".urdf" file. How can you run the controller successfully without  "go1.urdf"?

suihu-mk, 2022-10-01:
> > When I modify the robot.xacro file and generate the urdf file, it cannot run successfully.
> 
> What kind of modification?

<xacro:leg prefix="RF" mirror="-1" mirror_dae="False" front_hind="1" front_hind_dae="True">
        <origin rpy="0 0 0" xyz="${leg_offset_x} ${-leg_offset_y} 0"/>

Here, I modified “prefix” and "RF" because the robot.xacro file downloaded from unitreerobotics/unitree_ros is not defined the same as here. I modified a lot of different writing.

suihu-mk, 2022-10-01:
> > [image] [image]
> 
> The simulation launch file requires a ".xarco" file, and the OCS2 requires a ".urdf" file. How can you run the controller successfully without "go1.urdf"?

I'm not sure how it runs successfully.

qiayuanliao, 2022-10-13:
Maybe you can ask #9 for help.

AndrewZheng-1011, 2023-01-31:
Hey @suihu-mk 

Perhaps I can be of help for integrating go1 into the legged control repository. I was able to successfully integrate go1 from the  [unitree go1 repo](https://github.com/unitreerobotics/unitree_ros/tree/master/robots/go1_description/xacro) into the legged control repository. You can check out my [fork page](https://github.com/qiayuanliao/legged_control/commit/675ed9cf8ca860b89d26c1d2859696034d1214e7), but currently I have not asked for a PR because my folder structure for my xacro files is not as clean as qiayuanliao due to the common folder and its contents not being quite the same as A1. 

If you perhaps want to see some of the lower-level details, I have taken some notes while I was trying to integrate go1 in txt file attached below:
[integrating_go1.txt](https://github.com/qiayuanliao/legged_control/files/10549286/integrating_go1.txt)

Here is also a copy of the go1.urdf file ([go1.txt](https://github.com/qiayuanliao/legged_control/files/10549302/go1.txt)), but I believe the xacro file is necessary to launch the program.

And then lastly, here are some of the working results with the corresponding changes. 

[image]
[image]

AndrewZheng-1011, 2023-02-21:
Hello @qiayuanliao 

Regarding integrating unitree go1. I am having some tf warning when directly integrating go1 into this framework. Regarding changes to unitree go1 xacro files, the major changes I made were:
- naming conventions (e.g. link names: FR -> RF, _hip_joint -> _HAA, etc.)
- A few plugin switches (e.g. controller plugin: libgazebo_ros_control.so -> liblegged_hw_sim.so, adding p3d_base_controller, etc.)
Let me know if you need a more thorough list.

Essentially, when running the controller, I receive a repeated warning that the TF has redundant timestamps
```
...
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame LH_calf at time 130.876000 according to authority unknown_publisher at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp   
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame RF_hip at time 130.876000 according to authority unknown_publisher at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp 
...                                           
```
These TF_REPEATED_DATA warning are listed for all links.

When I do not load the legged controller plugin, these warnings evidently disappear (as joint state TF frames are most likely not being published). I am wondering if you could help me debug or find a quick solution to the redundant frames before I look into the controller and estimation plugin/source code.

I believe you can run my version of the code [here](https://github.com/AndrewZheng-1011/legged_control/tree/ece97919b14fff0b202855f8a378b7e93c4b0311). This version should have the same command as yours:
```
export ROBOT_TYPE=go1
roslaunch legged_unitree_description empty_world.launch robot_type:=go1
roslaunch legged_controllers load_controller.launch cheater:=false
```
```
rosservice call /controller_manager/switch_controller "start_controllers: ['controllers/legged_controller']                   
stop_controllers: ['']
strictness: 0
start_asap: false
timeout: 0.0" 

```

qiayuanliao, 2023-02-21:
I got the same warning in my original controller and boarded me a lot too. I guess it's cause by LeggedRobotVisualization in ocs2_legged_robot_ros. However, I don't have time to dig deeper about this warning. I will be very appropriate if you can find out what's the problem.

AndrewZheng-1011, 2023-02-22:
I'll see if I can take a look into it. Not sure the timeline with my schedule as well, but I will give an attempt at it.

qiayuanliao, 2023-05-04:
> I'll see if I can take a look into it. Not sure the timeline with my schedule as well, but I will give an attempt at it.

I find out that in some high-performance computers, there is no such problem, so strange....

RamCharanThota, 2023-06-06:
> Hey @suihu-mk
> 
> I have been successful in integrating go1 into the legged control repository and perhaps my experience can help close this issue page. I integrated go1 from the [unitree go1 repo](https://github.com/unitreerobotics/unitree_ros/tree/master/robots/go1_description/xacro) into the legged control repository which can be seen in this [fork page](https://github.com/clemson-dira/legged_control/tree/unitree_go1), but currently I have not asked for a PR because my folder structure for my xacro files is not as clean as qiayuanliao due to the common folder and its contents not being quite the same as A1.
> 
> If you perhaps want to see some of the lower-level details, I have taken some notes while I was trying to integrate go1 in txt file attached below: [integrating_go1.txt](https://github.com/qiayuanliao/legged_control/files/10549286/integrating_go1.txt)
> 
> Just to highlight some of the details:
> 
> 1. Change dae path files when transferring unitree go1 files into this repo
> 2. Replace unitree's imu w/ legged_control's imu.xacro
> 3. Notation name changes. (FL_hip -> LF_hip)
> 4. Integrate legged_control's controller and state estimator plugin (liblegged_hw_sim.so & p3d_base_controller)
> 
> Here is also a copy of the go1.urdf file ([go1.txt](https://github.com/qiayuanliao/legged_control/files/10549302/go1.txt)), but I believe the xacro file is necessary to launch the program.
> 
> And then lastly, here are some of the working results with the corresponding changes.
> 
> [image] [image]

Hi @AndrewZheng-1011 . I am trying to control my custom robot using this package. I am wondering how did you create gait.info reference.info and task.info files for your robot model. I really appreciate if you can help with this information.

fangzhiyuan1995, 2023-06-06:
> > I'll see if I can take a look into it. Not sure the timeline with my schedule as well, but I will give an attempt at it.
> 
> I find out that in some high-performance computers, there is no such problem, so strange....

[image]
Although the error has been corrected and disappeared, sometimes ros:: time is more important.

fangzhiyuan1995, 2023-06-06:
> > I'll see if I can take a look into it. Not sure the timeline with my schedule as well, but I will give an attempt at it.
> 
> I find out that in some high-performance computers, there is no such problem, so strange....

[image]
Although it can solve the above problem, sometimes ros:: time is more important

nanbwrn, 2023-10-22:
hi,@AndrewZheng-1011 
I ran the 'go1' package you provided, and after running the 'roslaunch legged_unitree_description empty_world.launch' command, I noticed that there is no robot model in the opened Gazebo. The compilation has already been successful, and I've set the environment variables. Here is a video

https://github.com/qiayuanliao/legged_control/assets/132453617/6b61c4cc-d68e-4bca-885c-87d0f735b999

This warning appeared in the terminal.
Warning [parser.cc:950] XML Element[iters], child of element[physics] not defined in SDF. Ignoring[iters]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.

How can I make the robot model display correctly? Is it because I haven't generated the URDF file? I noticed there's a URDF generation script (generate_urdf.sh). How do I correctly run this script?

AndrewZheng-1011, 2023-10-22:
> hi,@AndrewZheng-1011 I ran the 'go1' package you provided, and after running the 'roslaunch legged_unitree_description empty_world.launch' command, I noticed that there is no robot model in the opened Gazebo. The compilation has already been successful, and I've set the environment variables. Here is a video
> 
>  Gazebo.Erro.mp4 
> This warning appeared in the terminal. Warning [parser.cc:950] XML Element[iters], child of element[physics] not defined in SDF. Ignoring[iters]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
> 
> How can I make the robot model display correctly? Is it because I haven't generated the URDF file? I noticed there's a URDF generation script (generate_urdf.sh). How do I correctly run this script?

I changed the launch file structure in mine so you can simply use the following command instead:
```
roslaunch legged_unitree_description load_world.launch
```

I believe the rest is simply the same as the original repository.

nanbwrn, 2023-10-31:
@AndrewZheng-1011 Sorry for the delayed response, thank you.

qiayuanl, 2023-11-12:
Go1 support is added and tested in the real robot.

closed 2023-11-12 by qiayuanl