qiayuanl/legged_control #12: A1 robot lie down on the ground, not stand up in gazebo.

JACKLiuDay, 2022-11-11:
Hi, thank you for your excellent work about this project. I am trying to learn how to use this project for simulation. When I launch the empty_world.launch. The a1 robot just lie down in gazebo. It does not stand up like ANYmal in OCS2. When I type trot, the robot can trot but still not stand up. Could you give me some advice on how to let a1 stand up and walk in gazebo. Thank you so much.
[image]

qiayuanliao, 2022-11-11:
Make sure you have read the README carefully! Check [this section](https://github.com/qiayuanliao/legged_control#note).

closed 2022-11-11 by qiayuanliao

JACKLiuDay, 2022-11-11:
> Make sure you have read the README carefully! Check [this section](https://github.com/qiayuanliao/legged_control#note).

Hi, sorry, I am rookie to ROS and this project. I got the reason now and a1 can stand up now. Thank you.

Kashu7100, 2022-11-11:
Could you tell me how to add `cmd_vel` slider in Rviz?

Thank you

qiayuanliao, 2022-11-12:
> Could you tell me how to add `cmd_vel` slider in Rviz?
> 
> Thank you

[rqt_robot_steering](http://wiki.ros.org/rqt_robot_steering)

BolunDai0216, 2023-08-21:
For future users, you need to run

```bash
rosrun rqt_robot_steering rqt_robot_steering
```

to open the `rqt_robot_steering` GUI

Buddy-X, 2025-05-08:
> > Make sure you have read the README carefully! Check [this section](https://github.com/qiayuanliao/legged_control#note).
> 
> Hi, sorry, I am rookie to ROS and this project. I got the reason now and a1 can stand up now. Thank you.

Hi, I’m currently facing the same issue as you did. Could you please share how you solved it? Thank you so much for your help!