qiayuanl/legged_control #13: How to use this project for my own robot?

JACKLiuDay, 2022-11-16:
Hi, thank you for sharing your project legged_control. I tried it and this project is brilliant. Now I have my own quadruped robot and xacro file.  I want to use this framework to realize the simulation of my own robot. In addition to replacing xacro, are there any control parameters that need to be adjusted? Thank you.

qiayuanliao, 2022-12-15:
Recommend process: 
- Leave my code as a dependency, completely alone;
- Add a new ros package, named `my_robot_descirtion`, imitate the name, and launch files of [unitree_description](https://github.com/qiayuanliao/legged_control/tree/master/legged_examples/legged_unitree/unitree_description);
- Test in the Gazebo by running the new empty_world.launch in `my_robot_descirtion`;
- Add a new ros package, named `my_robot_hw`, imitate the name, code, and files of [unitree_hw](https://github.com/qiayuanliao/legged_control/tree/master/legged_examples/legged_unitree/unitree_hw);
- Test `my_robot_hw` in the real world.

JACKLiuDay, 2022-12-22:
> Recommend process:
>
>     * Leave my code as a dependency, completely alone;
>
>     * Add a new ros package, named `my_robot_descirtion`, imitate the name, and launch files of [...]

Hi, thank you for your reply. I just tried your way. The xacro description file was added. And I tested it in the Gazebo. The robot showed well. 
[image]
And I want to use the simulation robot in Gazebo just like the project. When I run the

qiayuanliao, 2022-12-22:
I believe that you don't need to change any parameters if you correctly write the urdf.  Maybe you should check whether yoru zero position and direction of joints are the same as my urdf.