qiayuanl/legged_control #14: cann't switch legged_controller  when apply on hardware?

zhuiyuehun, 2023-01-09:
Hi, thank you for sharing your project legged_control，Follow the instructions, running legged_unitree_hw.launch on computer, but when use rqt_controller_manager to start the controller, the rqt_program stucked everytimes. what should i do can fix it ?

zhuiyuehun, 2023-01-09:
Using rqt_controller_manager to start the controller is success in gazebo

qiayuanliao, 2023-01-12:
Could you make sure you are using the last code? This problem was fixed in [this commit](https://github.com/qiayuanliao/legged_control/commit/061059a51c22a1702340fea4f1e86a5bf30dd4fe) two weeks ago.

qiayuanliao, 2023-02-10:
Sorry, I made a mistake. This problem should be solved in [this commit](https://github.com/qiayuanliao/legged_control/commit/0da28d6222219e597fbda2a26606239919b7a034).

closed 2023-02-10 by qiayuanliao