qiayuanl/legged_control #16: Unitree Go1 is leaning

abrobotics, 2023-01-23:
Hello, thank for this repository, it helps a lot!

I have a problem when I launch the controller on Unitree Go1:
I robot is not flat, and when I trot, the rear right (RR) hip seems to be lower that the others.
the most shocking behavior is that the robot limps to the right whereas the command is simply forward or backward ([you can see in this video](https://drive.google.com/file/d/1TCNN8G3DWZuU-PmXCq7w8Z9Uj9FvxR9b/view?usp=sharing)).

You can see that the RR foot force is different from the others:
[image]
[image]

Can you help with this ? I don't know if it's from IMU covariance, some hardware flaws or IMU/Joint bad calibration maybe.

qiayuanliao, 2023-01-24:
I'm happy that you like this repo. Your figure and plots are very convenient for me to find and solve your problems.

I have encountered some issues like yours in the A1 platform. The problem could be solved by joint calibration.

My A1 isn't calibrated so well when out-of-shell that it's visible to the naked eye (movement asymmetry even with the built-in controller).

You can communicate with the engineer in Unitree Robotics, who will tell you how to recalibrate the joint.

abrobotics, 2023-01-25:
Thank you, I'll reach them

elpimous, 2023-02-19:
@abrobotics, je ne serai pas contre un peu d'aide, si vous avez 2 secondes 😋
https://github.com/qiayuanliao/legged_control/issues/19#issue-1590848294

liukaishu, 2023-02-20:
@elpimous Hello, have you solved this problem?

elpimous, 2023-02-20:
@liukaishu Hey, not yet. Mr qiayuanliao is helping me, cool. You are welcome to help too. Thanks

zhuiyuehun, 2023-02-22:
@abrobotics @elpimous hello,brother,have u solved this problem?

elpimous, 2023-02-22:
Hey, @zhuiyuehun 
sorry, I'm not at this point for now, sorry

AndrewZheng-1011, 2023-04-05:
Hey,

I would like to get some information on the issues in the leaning of the robot. I did these in simulations such that there would be no hardware problems.

**_Stance Leaning_**
So, I noticed that that both go1 and a1 robot has some leaning issues given enough time. For example, when letting the robot remain in stance for long periods of time, the robot starts to lean. Here are some pictures to help visualize the problem:

**A1 Robot stance $t \geq 60s$**
[image]

**Go1 Robot stance $t \geq 60s$**

[image]

**_Trot Drift_**
Lastly, I am not sure if this error propagates toward when the robot is in stance, but, I conducted an experiment in which the robot starts to trot at the origin of the gazebo world. After $t\geq60s$, the robot starts to drift from the origin. Here are some more picture to help visualize the problem:

**Go1 trot at $t=0s$**
[image]

**Go1 trot at $t\geq60s$**
[image]

**Question**
Could I get some insights into what may be causing this? Is there a way to fix the drifting or leaning issue?

Peterxjerk, 2023-04-12:
> Hello, thank for this repository, it helps a lot!
>
> I have a problem when I launch the controller on Unitree Go1: I robot is not flat, and when I trot, the rear right (RR) hip seems to be lower that [...]

Hello, I am also trying to apply this controller on Unitree Go1. Can you give me some suggestion on how to modify this package to make it suitable for Go1 robot? Should I change the unitree_sdk which is suitable for Go1?

lucasbrodo, 2023-10-10:
Hi @AndrewZheng-1011,

Have you found a way to improve the trot drifting ?

Thanks for the feedback.

lucasbrodo, 2023-10-13:
Hi all,
So I have changed in `task.info` the hip angles and set them to 0.00. Looks nicer when the robot is standing.

However, it didn't affect the drifting. I tried to modify the MPC weight matrix Q to add more penalty on the x and y positions but no luck with that.
I have also tried to increase the `imuProcessNoisePosition`and `imuProcessNoiseVelocity`. No difference. 
I have also tried the joint calibration but no effect whatsoever.

The robot keeps drifting backwards and ends up falling on its own.

@elpimous @AndrewZheng-1011 @abrobotics Any ideas guys ?

Thank you very much for your help !

elpimous, 2023-10-13:
@lucasbrodo
Hello Lucas.
Could you help here ?
https://github.com/qiayuanliao/legged_control/issues/51#issue-1920700311
Imu is working, motors are moving correctly, with 10Nm.
I dont anderstand why robot can not maintain it position without trembling...
I dont know where to start searching. 😕

abrobotics, 2023-10-14:
Hi all,

Calibrating joints did help but there was still some stance leaning and trot drift. I managed to operate the Go1 robot during trials, but the motion still wasn't perfect. 

Unfortunately, I finished my work on this project and I am no longer working on Unitree Go1 robot.
As soon as I will get back access to my repo, I will share it here !

Thank's yall and good luck :)

YifuYuan, 2024-01-23:
@lucasbrodo 
Hi Lucas, did you eventually find a way to solve the trot drifting issue? We implemented the legged_control framework on our Go1 and had the same issue. We've also tried the joint calibration but it didn't help. Appreciate any advice!

AnudeepSajja, 2025-01-03:
I faced the same issue while implementing the framework on go1, however the issue for me is the foot force threshold, as the threshold is not correctly set the state estimation is facing some issue. I would suggest to change the threshold from40, so that the contact can be detected and the state estimation would work fine which avoids this drift.