qiayuanl/legged_control #17: Weighted-WBC VS HOQP-WBC

Harvey2019, 2023-02-02:
Hi, What a nice work!
Here I want to ask a question about the WBC. In your code, I see two types of WBC: Weighted one and HOQP, and you use the former in the code. So I want to know if the Weighted-WBC is more convenient or something? And is there obvious difference of performance between them?
Thank you very much!

qiayuanliao, 2023-02-03:
According to "Perceptive Locomotion through Nonlinear Model Predictive Control":

> While we have used a hierarchical resolution of tasks in the past [91], in this work, we instead use a single QP and trade off the tracking tasks with weights. We found that a strict hierarchy results in a dramatic loss of performance in lower priority tasks when inequalities constraints are active. Additionally, the complexity of solving multiple QPs and null- space projections in the hierarchical approach is no longer justified with the high quality motion reference coming from the MPC.

where [91] is “Perception-less terrain adaptation through whole body control and hierarchical optimization."

In addition, the weighted WBC is 25% faster than the hierarchical WBC in my test.

Harvey2019, 2023-02-04:
Thanks for your answer! I got it!

closed 2023-02-04 by Harvey2019