qiayuanl/legged_control #19: Add imu subscriber for hardware interface READ() function

elpimous, 2023-02-19:
Hello.
I'm impressed with your controller. Think it's exactly what I need for my real robot YLO2.
Tested lots of controllers (champ, Wolf, quad-sdk...) 

However don't have succes when trying to integrate imu subscriber on the code, to be able to call imu message into read() function of the robot hardware interface.

A1 has imu integrated in it lib, and some of my friends and I need to call external topic imu.

Could you add the imu subscriber integration in legged_unitree_hw code, for example?
Perhaps it could extend users and give big help for us 😉

Thanks qiayuanliao.
Vincent.

qiayuanliao, 2023-02-20:
Hi, I am happy that you enjoy this controller. Can you show me your code? I believe that it's easy to add an IMU subscriber on the HardwareInterface, and you may have some bug. I can try to help you find out.

elpimous, 2023-02-20:
Hello. What a quick answer...
humm i can't call imu message under read() function. here is what i'd like to do :

`void Ylo2HW::read(const ros::Time& /*time*/, const ros::Duration& /*period*/) {
  //udp_->Recv();
  //udp_->GetRecv(lowState_);

  for (int i = 0; i < 12; ++i) {
    jointData_[i].pos_ = 0.0; //
    jointData_[i].vel_ = 0.0; //
    jointData_[i].tau_ = 0.0; //
  }

  // TODO need to read here imu message.
  // imu_message contain the real imu topic msg
  imuData_.ori_[0] = 0.0; // imu_message.orientation.x;
  imuData_.ori_[1] = 0.0; // imu_message.orientation.y;
  imuData_.ori_[2] = 0.0; // imu_message.orientation.z;
  imuData_.ori_[3] = 0.0; // imu_message.orientation.w;

  imuData_.angularVel_[0] = 0.0; // imu_message.angular_velocity.x;
  imuData_.angularVel_[1] = 0.0; // imu_message.angular_velocity.y;
  imuData_.angularVel_[2] = 0.0; // imu_message.angular_velocity.z;

  imuData_.linearAcc_[0] = 0.0; // linear_acceleration.x;
  imuData_.linearAcc_[1] = 0.0; // linear_acceleration.y;
  imuData_.linearAcc_[2] = 0.0; // linear_acceleration.z;`

elpimous, 2023-02-25:
thanks for answer.
Well, compilation is ok, but i can't read imu message ?!

As you said, I inserted imu callbacy into init() : https://github.com/elpimous/Legged_Control/blob/6a9354b88e7e46af02bfecea03d2d4d7ee3e1c12/legged_unitree_hw/src/UnitreeHW.cpp#L33

here there is the callback : https://github.com/elpimous/Legged_Control/blob/6a9354b88e7e46af02bfecea03d2d4d7ee3e1c12/legged_unitree_hw/src/UnitreeHW.cpp#L54-L59

Could you help in imu read example value here ? https://github.com/elpimous/Legged_Control/blob/6a9354b88e7e46af02bfecea03d2d4d7ee3e1c12/legged_unitree_hw/src/UnitreeHW.cpp#L105

@qiayuanliao Thanks a lot.
Vincent

closed 2023-05-12 by elpimous