qiayuanl/legged_control #21: Question about deploying on legged robot with different configuration

J0keryang, 2023-02-21:
I tried to understand the code in this project with the purpose of individual dissertation. But the legged model I designed is different from the Unitree A1. For each leg of Unitree A1, the sequence of robotic joint is HAA(thigh)--HEF(hip)--KFE(knee) but the sequence of my model is HFE(hip)--HAA(thigh)--KFE(knee）. I have modified the order of names (in the form of String) related to each joint of this project and tested my model in gazebo. As a result, it seems like my robot tended to move along y-axis (going forward means moving along x-axis). I am not sure how to modify the code in this project to solve this problem.

I'll be patient to wait for any solution. Thank you.

elpimous, 2023-02-21:
Hello @J0keryang 
My joints order is different too. 
I use a motor_adapter index to select joint, direction and controller ID, regarding to needed index. 
https://github.com/elpimous/Legged_Control

Hope it will help you.

Vincent

J0keryang, 2023-02-22:
> Hello @J0keryang My joints order is different too. I use a motor_adapter index to select joint, direction and controller ID, regarding to needed index. https://github.com/elpimous/Legged_Control
> 
> Hope it will help you.
> 
> Hummm did you change rotation order of joints into urdf? Haa is 1 0 0 Hfe and kfe are 0 1 0
> 
> Vincent

Thank you for your suggestion, Vincent. My friend found this problem might be caused by the difference between the source code of Qingyuanliao and Unitree SDK, in terms of the joint index. As you mentioned, we indeed take the same rotation order of joint as you did (Haa is 1 0 0 Hfe and kfe are 0 1 0). Also, I have marked your projects as great learning resources.

elpimous, 2023-09-18:
@J0keryang  please, up.