qiayuanl/legged_control #22: could not retrieve one of the required parameters: loop_hz or cycle_time_error_threshold or thread_priority

elpimous, 2023-02-23:
hello the team.
I followed A1 hw code, and call actual A1 description (for tests), but I encounter this error : 

```
roslaunch ylo2_legged_hw ylo2_legged_hw.launch 
... logging to /home/ylo2/.ros/log/e194df8e-b3cd-11ed-a199-d3492bcbad75/roslaunch-ylo2-ROG-Strix-G713QR-G713QR-51688.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ylo2-ROG-Strix-G713QR-G713QR:34953/

SUMMARY
========

CLEAR PARAMETERS
 * /ylo2_legged_hw/

PARAMETERS
 * /legged_robot_description: <?xml version="1....
 * /legged_unitree_hw/contact_threshold: 40
 * /legged_unitree_hw/cycle_time_error_threshold: 0.002
 * /legged_unitree_hw/loop_frequency: 500
 * /legged_unitree_hw/power_limit: 4
 * /legged_unitree_hw/thread_priority: 95
 * /robot_type: a1
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /
    ylo2_legged_hw (ylo2_legged_hw/ylo2_legged_hw)

auto-starting new master
process[master]: started with pid [51698]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e194df8e-b3cd-11ed-a199-d3492bcbad75
process[rosout-1]: started with pid [51708]
started core service [/rosout]
process[ylo2_legged_hw-2]: started with pid [51716]
[ERROR] [1677193236.036131777]: could not retrieve one of the required parameters: loop_hz or cycle_time_error_threshold or thread_priority
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not retrieve one of the required parameters: loop_hz or cycle_time_error_threshold or thread_priority
[ylo2_legged_hw-2] process has died [pid 51716, exit code -6, cmd /home/ylo2/catkin_ws/devel/lib/ylo2_legged_hw/ylo2_legged_hw __name:=ylo2_legged_hw __log:=/home/ylo2/.ros/log/e194df8e-b3cd-11ed-a199-d3492bcbad75/ylo2_legged_hw-2.log].
log file: /home/ylo2/.ros/log/e194df8e-b3cd-11ed-a199-d3492bcbad75/ylo2_legged_hw-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

elpimous, 2023-02-25:
I inserted my hw code into legged_unitree_hw, and it works without this error !

qiayuanliao, 2023-02-26:
The namespace of your node and your parameters is different.