qiayuanl/legged_control #29: contactState_ needed for Legged Control ???

elpimous, 2023-03-06:
Hello, is those variables really necessary for the controller?

https://github.com/qiayuanliao/legged_control/blob/eff523ddec6b392414073d7ff5f1534cd411c533/legged_examples/legged_unitree/legged_unitree_hw/src/UnitreeHW.cpp#LL57C3-L59C4

My robot lib doesn t have it.. 
Thanks
Vincent

qiayuanliao, 2023-03-07:
Yes, it is needed for this project. However, you can hack and use the MPC output as the contact state. I may write some code for this feature in the future.

elpimous, 2023-03-07:
Ohhh bad for me 😭
Thanks friend for the quick answer.
Now my robot code is nearly ready.
I tested a standup, and motors jiiters.
I am missing those contactstates.
Can I use an existing variable?
Will you have time, very soon, to code the mpc hack you said?
Have a nice day.

elpimous, 2023-03-19:
hi, @qiayuanliao , did you have time to add MPC output as the contact state ?
Thanks.

elpimous, 2023-03-25:
small up ?!

elpimous, 2023-03-28:
Hi Jim. Any hope for any mpc contact states?
See you.
Vincent

qiayuanliao, 2023-04-05:
Sorry for the dealy. I am working on another project currently. I will add this feature within a month.

elpimous, 2023-04-05:
Ohh happy for you, my friend. Wish you the best.
And don't forget me/us, please 😉
I'll wait necessary time. 🙏

LifelongYuan, 2023-04-06:
There is a quick way to hack:
Replace the [ line ](https://github.com/qiayuanliao/legged_control/blob/d2cc872693469e6b7d1b084b0fd568d6ee219879/legged_controllers/src/LeggedController.cpp#L159) with `contactFlag[i] = leggedInterface_->getSwitchedModelReferenceManagerPtr()->getContactFlags(currentObservation_.time)[i]`
So the state estimator directly acquires gait schedule instead of contact sensors.

elpimous, 2023-04-06:
Ohhh thanks a lot...
I'll test it tonight... I'm impatient. 👍

elpimous, 2023-04-07:
hello @LifelongYuan 
well, tested, and I can't standup. too much motors trembling (bad kp, kd ?)

here is the terminal error : 

```
...
[CppAdInterface] Loading Shared Library: /tmp/legged_control/a1/RH_FOOT_position/cppad_generated/RH_FOOT_position_lib.so
[CppAdInterface] Loading Shared Library: /tmp/legged_control/a1/RH_FOOT_velocity/cppad_generated/RH_FOOT_velocity_lib.so
[CppAdInterface] Loading Shared Library: /tmp/legged_control/a1/RH_FOOT_orientation/cppad_generated/RH_FOOT_orientation_lib.so
WARNING: Failed to set threads priority (one possible reason could be that the user and the group permissions are not set properly.)
WARNING: Failed to set threads priority (one possible reason could be that the user and the group permissions are not set properly.)
WARNING: Failed to set threads priority (one possible reason could be that the user and the group permissions are not set properly.)
[ WARN] [1680904568.708592601]: The root link base has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ WARN] [1680904583.005091446]: Controller Manager: To switch controllers you need to specify a strictness level of controller_manager_msgs::SwitchController::STRICT (2) or ::BEST_EFFORT (1). Defaulting to ::BEST_EFFORT.
[ INFO] [1680904583.006608700]: Waiting for the initial policy ...
[ INFO] [1680904583.034767466]: Initial policy has been received.
[ WARN] [1680904583.036258342]: Cycle time exceeded error threshold by: 0.0259329s, cycle time: 0.034266262s, threshold: 0.005s
ERROR:  Maximum number of working set recalculations performed
ERROR:  Maximum number of working set recalculations performed
ERROR:  Maximum number of working set recalculations performed
ERROR:  Maximum number of working set recalculations performed
...
```

I did : 
```
roslaunch legged_unitree_hw legged_unitree_hw.launch (my modified controller)
roslaunch legged_controllers load_controller.launch cheater:=false
```
here is ok !

and when : 
```
rosservice call /controller_manager/switch_controller "start_controllers: ['controllers/legged_controller']                   
stop_controllers: ['']
strictness: 0
start_asap: false
timeout: 0.0" 
```
terminal writes => `ok: True`

Then, all the engines are shaking al lot, and i must then cut their power.

Any idea ?  a kp kd problem ? Where could i reduce them ?

Impatient to read you, 
Vicent.

elpimous, 2023-04-24:
Seems working.. thank you so much

closed 2023-04-24 by elpimous