qiayuanl/legged_control #34: WARNING: The MPC time-horizon is smaller than the MPC starting time

elpimous, 2023-04-12:
Hello the team,
robot wants to stand up, but cant !!
Any help : 
Ther terminal :
```
currentTime: 2.47148	 Controller finalTime: 1.002
WARNING: The MPC time-horizon is smaller than the MPC starting time
```

elpimous, 2023-04-12:
return of terminal : 
```
WARNING: The MPC time-horizon is smaller than the MPC starting time.
currentTime: 18.9098	 Controller finalTime: 1.002
WARNING: The MPC time-horizon is smaller than the MPC starting time.
currentTime: 18.9098	 Controller finalTime: 1.002
WARNING: The MPC time-horizon is smaller than the MPC starting time.
currentTime: 18.9098	 Controller finalTime: 1.002
WARNING: The MPC time-horizon is smaller than the MPC starting time.
currentTime: 18.9098	 Controller finalTime: 1.002
[ INFO] [1681325171.157292356]:  Write ---> joint ID: [6] ; pos: [0.053199] ; vel: [0.072389] ; fftorque: [2.494649] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.157653967]:  Write ---> joint ID: [4] ; pos: [-0.067456] ; vel: [-0.011383] ; fftorque: [-1.726983] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.157974827]:  Write ---> joint ID: [5] ; pos: [-0.439647] ; vel: [0.052897] ; fftorque: [5.203169] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.158285956]:  Write ---> joint ID: [3] ; pos: [-0.044735] ; vel: [-0.063882] ; fftorque: [-2.650366] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.158593372]:  Write ---> joint ID: [1] ; pos: [0.061044] ; vel: [0.006691] ; fftorque: [-1.416724] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.158897142]:  Write ---> joint ID: [2] ; pos: [0.407973] ; vel: [0.068385] ; fftorque: [5.334076] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.159198226]:  Write ---> joint ID: [12] ; pos: [-0.057478] ; vel: [-0.010046] ; fftorque: [2.832379] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.159496428]:  Write ---> joint ID: [10] ; pos: [-0.068710] ; vel: [-0.002338] ; fftorque: [-1.331165] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.159801292]:  Write ---> joint ID: [11] ; pos: [-0.438601] ; vel: [-0.005741] ; fftorque: [5.563870] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.160105497]:  Write ---> joint ID: [9] ; pos: [0.047299] ; vel: [-0.001570] ; fftorque: [-2.692181] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.160412716]:  Write ---> joint ID: [7] ; pos: [0.065409] ; vel: [0.008342] ; fftorque: [-1.348919] ; kp: [0.000000] ; kd: [3.000000]
[ INFO] [1681325171.160716182]:  Write ---> joint ID: [8] ; pos: [0.441709] ; vel: [0.002359] ; fftorque: [5.895403] ; kp: [0.000000] ; kd: [3.000000]
[ WARN] [1681325171.161053005]: Cycle time exceeded error threshold by: 0.178164s, cycle time: 0.182163620s, threshold: 0.002s
```

elpimous, 2023-04-12:
I seems that the robot can't standup, but i see legs movments, at low torque.
In rviz, i can see the 5 odom points move on one side, but not going down (anything like a standup try)

elpimous, 2023-04-19:
small UP, friends ?

elpimous, 2023-04-24:
doesnt encounter this error anymore, after reinstalled !

closed 2023-04-24 by elpimous

TiMo688, 2024-09-26:
Hello, I encountered the same issue as you (MPC runtime error). Could you please let me know how you eventually resolved it? Specifically, which software packages did you reinstall? I would greatly appreciate your help in clarifying this! I'm deploying on a Jetson Orin Nano, and I have already applied the necessary kernel patches and enabled runtime permissions, but the error persists.

elpimous, 2024-09-26:
Sorry Friend.
I an on another project for now. Waiting for a fully commented sdk version : for a generic robot, and not only specific one. 
Imu add, fully explained motor commands...
See you perhaps on ros2 version