qiayuanl/legged_control #35: how to test new imu ?

elpimous, 2023-04-12:
Hello, I have a myahrs+ imu, working, inserted into robot.
imu_message is well feeded.
topic print correct values.

how to test it under rviz ? The robot should move, when rotating real one, correct ?
rviz fixed_frame is "base"

in robot.xacro, i have imu_link and unitree_imu as follow : 
```

<!-- IMU -->
    <joint name="imu_joint" type="fixed">
        <parent link="base"/>
        <child link="imu_link"/>
        <origin rpy="0 0 0" xyz="0 0 0"/>
    </joint>

    <link name="imu_link">
        <inertial>
            <mass value="0.001"/>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <inertia ixx="0.0001" ixy="0" ixz="0" iyy="0.0001" iyz="0" izz="0.0001"/>
        </inertial>
        <visual>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <box size="0.001 0.001 0.001"/>
            </geometry>
            <material name="red"/>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <box size=".001 .001 .001"/>
            </geometry>
        </collision>
    </link>

    <xacro:IMU connected_to="base" imu_name="unitree_imu" xyz="0. 0. 0." rpy="0. 0. 0."/>
```

in legged_unitree_hw.launch, i have : 

```
<!-- Launch the imu node -->
    <node pkg="quad_imu" type="quad_imu" name="quad_imu">
        <param name="port" value="/dev/ttyACM0" />
        <param name="baud_rate" value="115200" />
        <param name="frame_id" value="imu_link" />
        <param name="parent_frame_id" value="base" />
    </node>
```

everywhere else, it is `unitree_imu`

I'd like to confirm the correct connection between my imu, and the controller.
Thanks

elpimous, 2023-04-24:
Hello all, @qiayuanliao ,
after lot of tests, now, robot tries to standup !
1/ it seems that some joints are inverted ?! (but all my real robot joints moves, produce same moves on rviz. so they seems normal, no ?) Read() and Write() have same joints direction !

2/ My imu is working, put perhaps no good frame or link ? Should the robot rotate on rviz, when i rotate it on real ? I doesn't actually ! What steps should i check ? (Actually, it sends to imu/data )
Thanks a lot

elpimous, 2023-04-28:
Please, any help, anyone ?

qiayuanliao, 2023-05-04:
The IMU should be tested separately, you should check the orientation, linear acceleration, and angular velocity output is correct by some visualization tool. The determination of correct IMU data is out of this repo's scope. I will only suggest that you can try [rviz_imu_plugin](http://wiki.ros.org/rviz_imu_plugin) and plotjuggle, and keep in mind the unit! Once your IMU works and is installed into the robot with the same axis alignment as the base frame, the robot in RViz should move the same as the real robot.

You should implement some interface to set the direction of each joint in your hardware interface in order to keep the actually joint direction the same as the definition of URDF(controller).

Don't send any command before you are sure that all the data input (imu, joint position, velocity, torque) is correct, otherwise, you may heart the robot or yourself!!! IS VERY DANGEROUS!

elpimous, 2023-05-04:
Hi @qiayuanliao  Really happy to read you.
My imu is well fixed, and checked under rviz, of course. +X forward, +Y left, +Z up.

`Don't send any command before you are sure that all the data input (imu, joint position, velocity, torque) is correct, otherwise, you may hurt the robot or yourself!!! IS VERY DANGEROUS!`
Thanks a lot for security advice !! I'm testing with very low torque (< 5 Nm)

`You should implement some interface to set the direction of each joint in your hardware interface in order to keep the actually joint direction the same as the definition of URDF(controller). `
**It means that rviz will reflect exact real leg move, correct ?**

`Once your IMU works and is installed into the robot with the same axis alignment as the base frame, the robot in RViz should move the same as the real robot` **Here is my problem !!! Thanks.**
Need to check parent/child and tf...