qiayuanl/legged_control #40: Can't trot stably when deploying legged_control to real A1 robot

P1terQ, 2023-05-04:
I use legged_control without any modification to control the real robot and the robot can't trot stably in reality. If i give it velocity command, it will lose balance. Here are the failure vedios in reality and rviz

https://user-images.githubusercontent.com/90088571/236162530-6e736bbf-f613-4828-9b60-9ed17999f165.mp4

https://user-images.githubusercontent.com/90088571/236162574-dc9beaff-7632-470f-b43d-4dc89b85a67a.mp4

However, everything works fine in simulation. I guess the reason may be the addtional weight of VLP-16 and rack. So i add an additional link(2kg) with respect to base link in robot.xacro. The controller seems to handle the weight perfectly unlike the case in reality. 
I also try to add the lidar link t0 jointNames in ocs2/ocs2_robotic_examples/ocs2_legged_robot/include/ocs2_legged_robot/common/ModelSettings.h, but it makes little difference. Both case in simulation(with and without additional weight) works perfectly.

https://user-images.githubusercontent.com/90088571/236162608-0e097cf8-6841-465d-b372-e5f691afa98a.mp4

Is there any reason why the performance of legged_control degrade in my case?

qiayuanliao, 2023-05-04:
The problem can be solved by adding your payload to the robot.xarco. Because the model used by controller are not match the real robot. 

The controller uses the URDF in /temp/legged_control and the Gazebo simulation uses the URDF in ros params /legged_robot_description. Typically, [hw.launch](https://github.com/qiayuanliao/legged_control/blob/master/legged_examples/legged_unitree/legged_unitree_hw/launch/legged_unitree_hw.launch#L4C3-L8) or [empty_world.launch](https://github.com/qiayuanliao/legged_control/blob/master/legged_examples/legged_unitree/legged_unitree_description/launch/empty_world.launch#L4-L8) use a script to keep them the same so if you change the robot.xarco, both the controller and the simulation will use the correct URDF, and you will NOT observer some behavior like your first video -- the robot will handle the weight of lidar perfectly.

> I also try to add the lidar link t0 jointNames in ocs2/ocs2_robotic_examples/ocs2_legged_robot/include/ocs2_legged_robot/common/ModelSettings.h, but it makes little difference. Both case in simulation(with and without modification in ModelSettings.h) works perfectly.

Adding the lidar link t0 jointNames in ModelSetting.h means nothing! You should not modify the code of OCS2 if you don't 
clearly understand what you are doing!

P1terQ, 2023-05-06:
Thanks for your reply! After I remove the payload, the robot performs much better.  I will try to  modify robot.xacro to make the URDF match with the real robot equipped with lidar.

However, I meet with the issue mentioned in #30 which i think is caused by CoM bias. Therefore, the way that control robot by /cmd_vel doesn't work well. I will try to figure out how to mitigate its impact.

closed 2023-05-06 by P1terQ