qiayuanl/legged_control #42: issue with spawning my custom robot in gazebo simulation

RamCharanThota, 2023-05-23:
Hi,
i am trying to use this package to control my robot model in gazebo. I have changed my robot link and joint names to match legged_unitree_description. But, I am facing the following error
[image]
I have observed that it is mainly due to custom gazebo ros plugin section of the urdf
[image]. I really appreciate if you can help me with this issue.
And also I am wondering whether I need to create custom hardware interfaces for my robot even though i am testing in gazebo simulation not in physical robot.

RamCharanThota, 2023-05-31:
Hi @qiayuanliao , I solved the above issue. If you dont mind, can you tell how can I creare gait.info reference.info and task.info files for my custom robot. Thank you

neozoxix, 2024-01-07:
> Hi, i am trying to use this package to control my robot model in gazebo. I have changed my robot link and joint names to match legged_unitree_description. But, I am facing the following error ![mpc_control_issue](https://private-user-images.githubusercontent.com/67779589/240149735-6aad22a9-a457-43ed-be12-ab579987fa52.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2MTEzMjAsIm5iZiI6MTcwNDYxMTAyMCwicGF0aCI6Ii82Nzc3OTU4OS8yNDAxNDk3MzUtNmFhZDIyYTktYTQ1Ny00M2VkLWJlMTItYWI1Nzk5ODdmYTUyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA3VDA3MDM0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5ZGZkZTAyMDBiNTYwZjUyNzE1M2YwMjMyNTNmMzljYjczNTY3OGQ4YTZhMmUwYmMyZmEwZjllYmQ5YzQyMzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.dy_fJtHL7vDX6VkoXZOZzevbd_ly_EkDcvQVJa8uH4g) I have observed that it is mainly due to custom gazebo ros plugin section of the urdf ![gazebo_ros_interface](https://private-user-images.githubusercontent.com/67779589/240149985-12ff6673-3ed6-43d3-a9ee-93400ce51421.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2MTEzMjAsIm5iZiI6MTcwNDYxMTAyMCwicGF0aCI6Ii82Nzc3OTU4OS8yNDAxNDk5ODUtMTJmZjY2NzMtM2VkNi00M2QzLWE5ZWUtOTM0MDBjZTUxNDIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA3VDA3MDM0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA5YzE3YzQ5Mzk1OWNhMDI0Y2ZhYWVjZTk3ZWRlYjIwYTVlN2EyNjFiNzIzNmE0ZThkYjg3ZjAwNDA4Y2VkZTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.SuCflvvKijh1pyQ964cDcmp0hLmOOgWh8nDck4FacfM). I really appreciate if you can help me with this issue. And also I am wondering whether I need to create custom hardware interfaces for my robot even though i am testing in gazebo simulation not in physical robot.

Sorry, can you tell me how to solve that issue ? Thank you

hanzhuchen666, 2024-02-21:
> Hi, i am trying to use this package to control my robot model in gazebo. I have changed my robot link and joint names to match legged_unitree_description. But, I am facing the following error [...]

I have the same issue and don't know how to solve it.