qiayuanl/legged_control #47: rviz无法显示机器人模型

1242713693, 2023-08-01:
[image]

juanvon, 2023-08-13:
检查一下模型是不是正确的，参数是什么。我记得是legged_robot_description

1242713693, 2023-08-13:
> 检查一下模型是不是正确的，参数是什么。我记得是legged_robot_description

[image]
   这个还是不行，能麻烦看一下你的参数设置吗？谢谢

USTCzbb, 2023-08-28:
For the issue of the robot model not loading, everyone can visit my homepage to check. I will inform everyone how to solve this problem. Remember to give me a little star, thank you all!

1242713693, 2023-08-30:
> 对于机器人模型无法加载的问题，大家可以访问我的主页进行检查。我会告诉大家如何解决这个问题。记得给我一颗小星星，谢谢大家！

你好，你的主页的图片无法打开！！！

USTCzbb, 2023-08-30:
你加我QQ  3079386061  我教你   我放假了  身边没材料，你拍照片给我说啥问题

passer-by-Wang, 2023-09-01:
> [image]
兄弟，我也遇到这个问题了，你解决了吗

1242713693, 2023-09-01:
> > [image]兄弟，我也遇到这个问题了，你解决了吗

你将map改为base ；将robot_description 改为legged_robot_description试一下。

passer-by-Wang, 2023-09-01:
[image]
还是有一部分没有正常显示

1242713693, 2023-09-01:
> [image]还是有一部分没有正常显示

你在终端把指令全都打开，最后一步打开rviz

passer-by-Wang, 2023-09-01:
还是这样

1242713693, 2023-09-01:
> 还是这样
选择一种步态运行起来试一下

passer-by-Wang, 2023-09-01:
一样的结果

1242713693, 2023-09-01:
> 一样的结果
那我没办法解决了

passer-by-Wang, 2023-09-01:
你是设置那两个参数就正常了吗

1242713693, 2023-09-01:
> 你是设置那两个参数就正常了吗
我的也不正常，我设置的机器人是A1，但RVIZ出来的是MIT的模型。

passer-by-Wang, 2023-09-01:
我把环境变量设置加在.bashrc文件里source了，所有终端都能使用那个环境变量，然后那两个参数我改了之后还是模型显示不全

1242713693, 2023-09-01:
> 我把环境变量设置加在.bashrc文件里source了，所有终端都能使用那个环境变量，然后那两个参数我改了之后还是模型显示不全

那这个我没办法帮你，我也搞不清除

passer-by-Wang, 2023-09-01:
我发现问题在于没有开控制器，只开了控制器管理的gui没有选择控制器并打开，但是我现在发现，它的初始状态是趴着的，我直接执行trot步态就是显得很奇怪

elpimous, 2023-09-01:
Hi guys, please could you post in english too..
Each post is interesting (im french😋) 
Thanks friends.

mitsui29, 2023-10-24:
> 我发现问题在于没有开控制器，只开了控制器管理的gui没有选择控制器并打开，但是我现在发现，它的初始状态是趴着的，我直接执行trot步态就是显得很奇怪

兄弟想问一下，问题解决了吗？

1242713693, 2023-10-25:
> > 我发现问题在于没有开控制器，只开了控制器管理的gui没有选择控制器并打开，但是我现在发现，它的初始状态是趴着的，我直接执行trot步态就是显得很奇怪
> 
> 兄弟想问一下，问题解决了吗？

在命令框中执行
rostopic pub -1 /cmd_vel geometry_msgs/Twist "linear:
  x: 0.0
  y: 0.0
  z: 0.2
angular:
  x: 0.0
  y: 0.0
  z: 0.0"
  应该可以站立起来

mitsui29, 2023-10-25:
> > > 我发现问题在于没有开控制器，只开了控制器管理的gui没有选择控制器并打开，但是我现在发现，它的初始状态是趴着的，我直接执行trot步态就是显得很奇怪
> > 
> > 
> > 兄弟想问一下，问题解决了吗？
> 
> 没解决，但是趴着直接开控制器给trot步态就能直接站起来

感谢回复哈，我用这个rqt-->Plugins-->Robot Tools-->Robot Steering-->点击一下stop-->拖动滑块控制机器人速度和转向，也可以运行不过是在gazebo上，而且给到trot这个步态，gazebo里面无法起来，而是趴着颤抖，给到速度之后就能起来了，对于rviz我后来尝试了一下在Plugins的视觉化那个里面打开了一个实验版本的rviz,虽然可以完整加载模型了，但是发现模型外观上还是有些区别，然后模型也是穿过了整个地平面，对于给定的目标点，模型在rviz里面不会移动，在gazebo里面会同步移动。而且我发现那些个跳跃的步态基本上很难运行成功，很不稳定，容易翻车。下面有三张截图，前两张是rviz和gazebo 的联动，但是rviz的模型不会移动。单独打开一个终端开rosrun rviz rviz 只能显示一部分模型，也没反应。
[image]
[image]
[image]

mitsui29, 2023-10-25:
> > > 我发现问题在于没有开控制器，只开了控制器管理的gui没有选择控制器并打开，但是我现在发现，它的初始状态是趴着的，我直接执行trot步态就是显得很奇怪
> > 
> > 
> > 兄弟想问一下，问题解决了吗？
> 
> 在命令框中执行 rostopic pub -1 /cmd_vel geometry_msgs/Twist "linear: x: 0.0 y: 0.0 z: 0.2 angular: x: 0.0 y: 0.0 z: 0.0" 应该可以站立起来

好的，感谢哈。我刚刚回复的那个内容，兄弟有什么建议吗？

JunJieChenpete, 2023-11-02:
about the problem you mentioned that the legged robot do not move in rviz, you should change the "Fixed Frame" value to "odom", not the "base", for the reason that the legged robot are fixed with the "base". 
some people mentioned that the model loaded wrong in rviz, I truely recommand that use rviz with cammand `rosrun rviz rviz`, the rviz plugins in rqt can not run properly in my computer.

mitsui29, 2023-11-03:
> about the problem you mentioned that the legged robot do not move in rviz, you should change the "Fixed Frame" value to "odom", not the "base", for the reason that the legged robot are fixed with [...]

Thx for ur replying! But i mentioned before, the problem is that model loaded wrong when i used rosrun rviz rviz. i could not find the fixed frame "odom". Also when i was trying to add the legged_ model_discreption, last time i can still get part of model, but this time i get some error infos like this
[image]...

chdJanbin, 2024-04-18:
> > > > 我发现问题在于没有开控制器，只开了控制器管理的gui没有选择控制器并打开，但是我现在发现，它的初始状态是趴着的，我直接执行trot步态就是显得很奇怪
> > > 
> > > 
> > > 兄弟想问一下，问题解决了吗？
> > 
> > 
> > 没解决，但是趴着直接开控制器给trot步态就能直接站起来
> 
> 感谢回复哈，我用这个rqt-->Plugins-->Robot Tools-->Robot Steering-->点击一下stop-->拖动滑块控制机器人速度和转向，也可以运行不过是在gazebo上，而且给到trot这个步态，gazebo里面无法起来，而是趴着颤抖，给到速度之后就能起来了，对于rviz我后来尝试了一下在Plugins的视觉化那个里面打开了一个实验版本的rviz,虽然可以完整加载模型了，但是发现模型外观上还是有些区别，然后模型也是穿过了整个地平面，对于给定的目标点，模型在rviz里面不会移动，在gazebo里面会同步移动。而且我发现那些个跳跃的步态基本上很难运行成功，很不稳定，容易翻车。下面有三张截图，前两张是rviz和gazebo 的联动，但是rviz的模型不会移动。单独打开一个终端开rosrun rviz rviz 只能显示一部分模型，也没反应。 ![2023-10-26 04-06-11 的屏幕截图](https://private-user-images.githubusercontent.com/95944211/278141118-9ae9fe2e-6dcc-40a5-bd7c-99d3c7f0dd75.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM0MjAzNjIsIm5iZiI6MTcxMzQyMDA2MiwicGF0aCI6Ii85NTk0NDIxMS8yNzgxNDExMTgtOWFlOWZlMmUtNmRjYy00MGE1LWJkN2MtOTlkM2M3ZjBkZDc1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE4VDA2MDEwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkxZThjOTE5YzVmMzY0ZTY2OTBmMmY1ZDkzMTdmZWI3MzAyMmY0NmEzNzIyNjVjOTNkNjIyM2JmNGVlMmE5YTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8u5JKLwMnUGe3mAYgX2OJCFL6bP1uaS_sBQhahWCknE) ![2023-10-26 04-06-21 的屏幕截图](https://private-user-images.githubusercontent.com/95944211/278141163-29b43a31-c79d-482a-950f-d173407a0363.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM0MjAzNjIsIm5iZiI6MTcxMzQyMDA2MiwicGF0aCI6Ii85NTk0NDIxMS8yNzgxNDExNjMtMjliNDNhMzEtYzc5ZC00ODJhLTk1MGYtZDE3MzQwN2EwMzYzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE4VDA2MDEwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI0NzhlYWEyYWZhY2QxYzhkNjAzOGJlMzEwNGM2OGRjNDQyMmM3NTgxMTYxZTk1ZDNjMmZkMzA2M2M1Mzc4YzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.YT3QbTc8JCwNwXw_8cQC2FKSO_xQOtVcTrEZbR82An0) ![2023-10-26 04-06-54 的屏幕截图](https://private-user-images.githubusercontent.com/95944211/278141252-4da412bd-3d67-432b-bcd5-0393b636118e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM0MjAzNjIsIm5iZiI6MTcxMzQyMDA2MiwicGF0aCI6Ii85NTk0NDIxMS8yNzgxNDEyNTItNGRhNDEyYmQtM2Q2Ny00MzJiLWJjZDUtMDM5M2I2MzYxMThlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE4VDA2MDEwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1NTA1ZDViZWEzY2FlMDZmNzUxZDRkZDhiNDAyMTA4OTYyZTg0NjY2ZjI5MGZhMzE2NDk4NjdkZWRhMDI5ZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uXolKYnCLkn6bHBTvef6pM2z1IULgnlOFQukS7QflWQ)

rviz中机器人不动是因为你设置的固定坐标系fix frame是机身base，要设置为odom里程计坐标系；
机器人模型不显示，只能显示collision这个问题我也没解决   orz

zhj3180, 2024-05-09:
模型加载不完全我是参考https://blog.csdn.net/m0_46448663/article/details/136271265这篇帖子，修改完记得在终端里先source devel/setup.bash，再rqt。
一直趴着的问题，给一个cmd/vel，机器人就站起来了。用的rqt里的robot_steering