qiayuanl/legged_control #51: Joints trembling at startup, on video

elpimous, 2023-10-01:
Hello @qiayuanliao
a kp kd problem ?
where could I correct those values ?

https://github.com/qiayuanliao/legged_control/assets/8529940/ca4d8ec3-5e38-4c19-8412-9fdfac9e6a38

elpimous, 2023-10-04:
@qiayuanliao @LoveThinkinghard any updade, please ?!
my send command is in radians/s position mode. 
pos value is real radian turn, from zero position.

demirciomer, 2023-10-10:
hello @elpimous  did you find any solution? same issue for us.

elpimous, 2023-10-10:
Hey, could you share any video ?
Perhaps could we share ?!
elpimous12@gmail.com.
(pour commands format, refresh, robot specs...) 
This framework is really impressive.
Hope we'll have success on it.

evan-brooks, 2023-10-11:
Looks like not enough torque?

elpimous, 2023-10-11:
Hey,.... Not silly...
I did tests with low maxtorque..
>3Nm.
Need to test. Thanks friend.

elpimous, 2023-10-13:
@demirciomer , what robot do you use ? I suspect any task.info bad compatibility param !

demirciomer, 2023-11-17:
@elpimous we have a custom robot like you, but its size is bigger. We increased the torque limit in urdf and operated the robot in standing mode. Tries to increase body height when we switch to the controller from position control. still no fix available.

elpimous, 2023-11-17:
@demirciomer
Hello.
Motors always trembling ??
You don t have foot sensors, correct ? If yes, how did you faked foot sensors values ?
If anyone of us two find solution, please don t forget the other 😉

demirciomer, 2023-11-17:
@elpimous  We dont have foot sensors. We are just looking for the force in Z and above treshold we set foot contact as true.  We are looking MPC observation topic for forces in Z direction.

elpimous, 2023-11-17:
And you have same reaction as me ?
Robot cant maintain sitted down position.
Correct ?

demirciomer, 2023-11-17:
We tried another thing. We stand up the robot with position control and implemented a controller switch which switches the controller from position to legged control. it tries to move up from standing position. foot contacts are all true no problem in foot contacts but body tries to go up. may be the state estimation problem or controller task.info parameter problem.

elpimous, 2023-11-17:
> just be sure about the torque limits in the urdf file for your case.
> increase them to 100nm etc.

Ill Try it for sure. Ill tell you. Thanks

demirciomer, 2023-11-27:
@elpimous  also check the motor controllers default parameters if there is an integral coefficient, make it zero,  that is also problem. We are still dealing with it  :+1:

elpimous, 2023-11-27:
@demirciomer.
Thanks friend.
My Ki is =0.

Could you confirm this : 

- It uses position mode, 
- position for me is Nan, 
- velocity is signed and in rad/s, 
- torque is fftorque unsigned, in Nm, 
- kp and kd are unsigned, in Nm/rad,
- stop is signed, and in rad/s

In my bldc controllers (Moteus),

demirciomer, 2023-11-28:
@elpimous yeah same parameters, we also tried to smooth the control signals with low pass filters, it keeps the position better but this is not a good practice. probably state estimation of both of us are problematic. It might be  a noise issue. Keep try to solve it.

demirciomer, 2023-11-28:
@elpimous  torque values should be signed?

yyagin, 2023-12-07:
Hi, @elpimous. In literature, position control is used only when the feet are in the swing phase. During the stance phase, only torque control is applied. Therefore, you should provide only feedforward values (pos_des, vel_des, kp, and kd, all set to zero) while the robot is in the stance phase. In contrast, during the swing phase, all desired variables, including feedforward values, should be sent. Perhaps that's the issue you're encountering. By the way, @demirciomer  and I are working together, and this fix seems to solve our issue. We're continuing to test it.

elpimous, 2023-12-07:
Hi @yyagin @demirciomer 
Nice.
after a French comprehension of your answer, you seem to use 2 different orders ?!
a Torque mode for stance, and a position mode for swing ?
How do you did that ? could you share a piece of your controller code ?
Don't know how to switch from one to other in process ?
Thanks to integrate me in your equation Lol.
Have a nice day, friends.

elpimous, 2023-12-08:
Hi @yyagin , @demirciomer 
Hello friends. Yes, this is my github package.
ok to test with only fftorque feeded.
cross fingers ! 
I'll tell you.
Good night friends

demirciomer, 2023-12-13:
@elpimous first you can try to send just ff_torque values to joints withouth any other parameters pos vel kp kd, make them zero and just send ff torques. This would prevent trembling behaviour. later add pos error times kp + vel error times kd to the ff torque to follow pos and velocity