qiayuanl/legged_control #53: The angle of the legs in the go1 simulation is wrong.

KCCT-Miyamoto, 2023-10-12:
Hello. @qiayuanliao 
unitree_description empty_world.launch

The leg angle of go1 in the simulator appears to be 90 degrees different from the initial position. Can you tell me how to improve this?

[image]

Freaky1122, 2023-12-29:
Have you solved this problem?I met the same question

LeonardoGMoraes, 2024-01-01:
Did someone solve this? I have the same problem.

linzuen, 2024-01-13:
You can try to modify the parameters of https://github.com/qiayuanl/legged_control/blob/master/legged_examples/legged_unitree/legged_unitree_description/urdf/common/leg.xacro#L189-L190  to 0.2

KCCT-Miyamoto, 2024-02-29:
Thanks!! I was able to run the simulation.

closed 2024-02-29 by KCCT-Miyamoto