qiayuanl/legged_control #55: When running the code with the Go1 robot, it was noticed that the two hind legs are splayed outward.

nanbwrn, 2023-10-31:
When I run the simulation with the Go1 robot, I've noticed that the robot's leg positions are normal when it's standing. However, when it starts walking with trot gait, I observe that its rear two legs splay outward as it moves. When I run it on the actual robot, this strange phenomenon also occurs.
How should I make modifications to address this issue? 
Thank you, everyone.

 Below is a simulation video for reference.

https://github.com/qiayuanliao/legged_control/assets/132453617/2cad5884-ba9a-4d5e-ae84-df9c6c70c2ea

demirciomer, 2024-01-14:
Hi, can you send the reference.info file of your robot?

nanbwrn, 2024-07-18:
targetDisplacementVelocity          0.25;
targetRotationVelocity              0.5;

comHeight                           0.30 ; Unitree Go1 Default Spec: 0.325 -> 0.31

defaultJointState
{
   (0,0)   0.00   ; LF_HAA | Original: -0.20
   (1,0)   0.72   ; LF_HFE | Original: 0.72
   (2,0)  -1.44   ; LF_KFE | Original: -1.44
   (3,0)   0.00   ; LH_HAA | Original: -0.20
   (4,0)   0.72   ; LH_HFE | Original: 0.72
   (5,0)  -1.44   ; LH_KFE | Original: -1.44
   (6,0)   0.00   ; RF_HAA | Original: 0.20
   (7,0)   0.72   ; RF_HFE | Original: 0.72
   (8,0)  -1.44   ; RF_KFE | Original: -1.44
   (9,0)   0.00   ; RH_HAA | Original: 0.20
   (10,0)  0.72   ; RH_HFE | Original: 0.72
   (11,0) -1.44   ; RH_KFE | Original: -1.44
}

initialModeSchedule
{
  modeSequence
  {
    [0]  STANCE
    [1]  STANCE
  }
  eventTimes
  {
    [0]  0.5
  }
}

defaultModeSequenceTemplate
{
  modeSequence
  {
    [0]  STANCE
  }
  switchingTimes
  {
    [0]  0.0
    [1]  1.0
  }
}