qiayuanl/legged_control #65: Build in A1 (nvidia tegra xavier (nvgpu)/integrated, armv8, ubuntu 18.04)

Zhihaibi, 2024-06-08:
Hi,

I am trying to compile hpipm on Arm (nvidia tegra xavier (nvgpu)/integrated, armv8, ubuntu 18.04 )

and meet these error: Unrecognized command line option '-m64' and '-mavx'

[image]

I have already switched to the GENERIC target in HPIPM  in Cmakelist.txt:

### Target architecture
#set(TARGET AVX CACHE STRING "Set CPU architecture target")
set(TARGET GENERIC CACHE STRING "Set CPU architecture target")

 and in makefile.rule, I modify to:
### Target architecture
#TARGET = AVX512
#TARGET = AVX
TARGET = GENERIC

But I still get this error, do you know how to solve it? Thank you very much!

Zhihaibi, 2024-06-08:
[image]

nanbwrn, 2024-07-13:
hi,@Zhihaibi  
I encountered the same issue as you. Have you resolved it? Thank you very much for your response.

Zhihaibi, 2024-07-13:
I guess that ocs2 is only tested in 20.04. Update the system to 20.04 and try again.

nanbwrn, 2024-07-13:
My system is 20.04.

nanbwrn, 2024-07-13:
I am using the Jetson AGX Orin with system version ubuntu 20.04, but I'm still encountering the same error as you. It's likely due to compiling for the ARM64 architecture. The code compiles fine on my laptop, but I encounter errors when compiling it on the Orin.

fangzhiyuan1995, 2024-07-13:
[image]

> Hi,
> 
> I am trying to compile hpipm on Arm (nvidia tegra xavier (nvgpu)/integrated, armv8, ubuntu 18.04 )
> 
> and meet these error: Unrecognized command line option '-m64' and '-mavx'
> 
> ![9630b789c645be070e45f001ce36885](https://private-user-images.githubusercontent.com/42597645/337863254-d23ad81a-e82b-4626-84c7-2733567e6718.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA4NzI0MDUsIm5iZiI6MTcyMDg3MjEwNSwicGF0aCI6Ii80MjU5NzY0NS8zMzc4NjMyNTQtZDIzYWQ4MWEtZTgyYi00NjI2LTg0YzctMjczMzU2N2U2NzE4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzEzVDEyMDE0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZkMWU5MTc3NjgzNTYwMDM1OGNlZGIzZTUzOWI3NjI2MmE2YTc4ODcyNzk3NmQxMTJlZjU3MmVjOTZmNWZkZWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hP0rpnEFPOoPWCAVG-nTzv96rRBRETtKNaxHmiTcxEs)
> 
> I have already switched to the GENERIC target in HPIPM in Cmakelist.txt:
> 
> ### Target architecture
> #set(TARGET AVX CACHE STRING "Set CPU architecture target") set(TARGET GENERIC CACHE STRING "Set CPU architecture target")
> 
> and in makefile.rule, I modify to:
> 
> ### Target architecture
> #TARGET = AVX512 #TARGET = AVX TARGET = GENERIC
> 
> But I still get this error, do you know how to solve it? Thank you very much!

set(TARGET GENERIC CACHE STRING "Set CPU architecture target")
At the same time, it is necessary to delete the compilation history of hpipm_catkin in the build and develop folders, and then compile the catkin build hpipm_catkin. The system uses Ubuntu 20.04