qiayuanl/legged_control #79: Using our own robot, the program can run, but the MPC spends a large amount of time on linearization.

Wan020607, 2025-03-18:
After using my own robot, I modified the **legged_robot_sqp.launch** file to run my robot and tested the MPC part. However, the robot had no response to the gait commands and kept issuing warnings:WARNING: The solution time window might be shorter than the MPC delay!
After that, I modified the task.info file to set printSolverStatistics to true in the SQP section to test the SQP solver's performance. The results are as follows:
The benchmarking is computed over 20 iterations. 
SQP Benchmarking           :    Average time [ms]   (% of total runtime)
        **LQ Approximation   :    1528.7 [ms]             (93.5049%)**
        Solve QP           :    8.35811 [ms]            (0.511233%)
        **Linesearch         :    97.5141 [ms]            (5.96456%)**
        Compute Controller :    0.315152 [ms]           (0.0192766%)
It was found that the SQP solver spent a large amount of time on **linearization**. I'm not sure what else needs to do besides changing the URDF, configuring the corresponding **task, reference, and gait files,** and updating **the names of the joints and links** in code.
Thank you for your response!

jiaweinie, 2025-05-17:
Hello, have you solved your problem? I have this problem.

[image]
I adjusted the mpc window time in the task，but it seems to have no effect
could you give me some views

Wan020607, 2025-05-18:
Did you encounter the problem during the execution of legged_robot_sqp.launch? I ran catkin clean and then rebuild it, and it worked successfully.

jiaweinie, 2025-05-18:
[image]
Thank you very much. I replaced my robot according to the Readme and this problem occurred when I started the controller. However, according to your instructions, this problem has disappeared. This is very strange. However, there is still a problem with qpoases, and I am solving it.